RespiraWorks/Ventilator #473: Less noisy/repetitive way of specifying integration tests in build system

jlebar, 2020-06-09:
The current way we're specifying integration tests in the build system is leading to noise and repetition.

I should say upfront, the way we're doing it seems to me very natural in PlatformIO, and I'm not sure how else we *should* do it.  So this isn't criticism of the authors of the code; it's a problem with PlatformIO.  But having said that, I still think we need to figure out something to do about it.

 - It's leading to noise because when you run `pio test -e native`, it prints every environment that it skipped:

    ```
    $ pio test -e native
    checksum        stm32                 IGNORED
    checksum        stm32-test            IGNORED
    checksum        stm32-dma-test        IGNORED
    checksum        int-test-blower       IGNORED
    checksum        int-test-solenoid     IGNORED
    checksum        int-test-stepper      IGNORED
    checksum        int-test-pinch-valve  IGNORED
    checksum        native                PASSED    00:00:02.539
    circ_buff       stm32                 IGNORED
    circ_buff       stm32-test            IGNORED
    circ_buff       stm32-dma-test        IGNORED
    circ_buff       int-test-blower       IGNORED
    circ_buff       int-test-solenoid     IGNORED
    circ_buff       int-test-stepper      IGNORED
    circ_buff       int-test-pinch-valve  IGNORED
    circ_buff       native                IGNORED
    ...
    ```

    We have an n^2 problem of integration tests times unit tests, that's going to become bad very quickly.

- It's leading to repetition in the `platformio.ini` file.  If and when we want to change flags in a way that affects integration tests, there is a high likelihood we're not going to do so in a consistent way, and ultimately that will lead to sadness.

cc @martukas

jlebar, 2020-06-09:
cc @Miceuz

martukas, 2020-06-10:
Yup, this is ugly :( I could not think of how to do it better. The problem is, I do want separate executables. I don't want to bring in the extra baggage of comms and debug libraries for when I'm only testing a couple of low level things. Do you think that's an unnecessary limitation?

jlebar, 2020-06-10:
> I don't want to bring in the extra baggage of comms and debug libraries for when I'm only testing a couple of low level things. Do you think that's an unnecessary limitation?

(Just as a point of order, if you're not using "the whole shebang", then it's confusing to me to call it an "integration test".  In the parlance I am familiar with, integration tests are whole-system tests.  I don't immediately have a name for this, but it's closer to a unit test than an integration test.  "Externally-validated unit test"?  I dunno.)

I am not sure what's the problem with compiling in e.g. the comms code even if you're not testing it.  In platformio we put most of our code into a single "core" library, because in my experience splitting up into many small libraries ends up being one big pan.  You have to avoid circular dependencies, which is painful, and then as soon as you have one you can't avoid you have to merge the libraries anyway.

But I'm also not sure how avoiding compiling in unneeded code relates to the problems from the first comment.

martukas, 2020-06-10:
I think we have had this set of definitions outlined by Jerome for a while now:
https://github.com/RespiraWorks/VentilatorSoftware/wiki/Embedded-software-testing
That's the terminology that I have been following. It makes sens to me, but perhaps there is something better? Feel free to propose.

In any case, whatever we call it, sure, it is much closer to unit tests, but with hardware.

My issue is not so much with compiling in the "whole shebang" as being dependent on say
* serial communications AND
* the debug interface AND
* some nested loops AND
* something else...
all running correctly when I want to test only the stepper, for example. If only there was a way to pass a single parameter to the controller loop as it starts...

What I am not aware of is maybe all of those systems I have listed above are thoroughly tested already? Then I could discard my concerns and do all this testing via the debug interface. I'm sincerely shopping around for alternatives.

jlebar, 2020-06-11:
> https://github.com/RespiraWorks/VentilatorSoftware/wiki/Embedded-software-testing

Whoa, one of us *wrote something down*?!

FWIW here's a citation for the definition of "integration test" that I'm familiar with. 
 https://en.wikipedia.org/wiki/Integration_testing  I wasn't able to find online anyone using the other definition, although I guess it's pretty hard to search for.

(This is a problem I have with the wiki -- since there's no notification or review process, there's no opportunity to get on the same page about things.)

Anyway I guess it's OK if it's written down!

> is maybe all of those systems I have listed above are thoroughly tested already?

Definitely not.

I think I understand your problem better, thanks.

A problem with this approach is that it only works inasmuch as the stuff you want to test is broken out into separate libraries (right?).  You got lucky that hal is a separate library, so I guess it works inasmuch as you are only testing hal.  If you are going to want to test stuff that's currently in core, we may need to come up with another approach, because, for the reasons explained in core/README.md, I don't think we should split up core into lots of tiny libraries.

But orthogonal to that, help me understand this: If you only want to depend on hal, how does that affect the issues I raised at the beginning of this bug?  How does "I only want to depend on hal" mean "I need a new PIO build environment for each integration test"? (lol integration test is still hard for me to say here.  I'll get to used to it.)

martukas, 2020-06-11:
Maybe I'm misunderstanding "environment". I somehow got the idea into my head that what pio sees as "environment" is what would equal "target" in the CMake-verse? Can I have one "environment" but create multiple executables for it, i.e. separate **main()** functions?

jlebar, 2020-06-11:
> I somehow got the idea into my head that what pio sees as "environment" is what would equal "target" in the CMake-verse?

I think pio conflates two concepts.  An environment is a target *and also* it's a set of build flags, a choice of toolchain, etc.

The reason is, they're primarily concerned with uploading stuff to an embedded device, where there really is ultimately only one binary.

> Can I have one "environment" but create multiple executables for it, i.e. separate main() functions?

You're not gonna like it, but I think one could hack it...  Basically you can use "advanced scripting" to do whatever you want.  Such as, read an environment variable and then set src accordingly, or set `-D` flags to the compiler.  Then you'd be able to do

```
TEST=foo pio run -e integration-test
```

and the TEST envvar could be interpreted by your script.  If the TEST envvar is not set, the script prints an error or whatever.

Beautiful?  No.  Scalable?  Possibly.

Miceuz, 2020-06-12:
Can't we split the integration test `env`s to a separate `platformion.ini` file and place it in a separate directory?

jlebar, 2020-06-12:
Not a bad idea!  If it gets repetitive we could even write a Python script to generate it.

martukas, 2020-06-26:
I think I took care of this with latest PR. There is now only one target. It is the "not pretty" way that @jlebar suggested, but at least I am satisfied with this and it solves the problem as defined here.
Are we satisfied to close this ticket?

jlebar, 2020-06-26:
Closed by https://github.com/RespiraWorks/VentilatorSoftware/pull/718

closed 2020-06-26 by jlebar