RespiraWorks/Ventilator #543: Building GUI on raspbian as per instructions does not work

martukas, 2020-06-17:
**What is wrong**
Fresh Raspbian system. I'm following the canonical instructions for building the GUI as described here:
https://github.com/RespiraWorks/VentilatorSoftware/blob/master/gui/README.md
Installed all the qt packages by copypasting from the same readme.
It does not build.

**How do you know it is wrong**
Not building = bad.
Good would be if it builds. This used to work.
Since we are still unsure of how performant and reliable our docker-based deployment is, we have to be able to build it manually.

**Code pointers**
Possibly new dependencies have been added.

Here is the error I get:
```
./gui.sh --build
++ dirname ./gui.sh
+ cd .
++ uname -s
+ PLATFORM=Linux
+ '[' Linux '!=' Darwin ']'
+ '[' Linux '!=' Linux ']'
+ '[' --build == --help ']'
+ '[' --build == --install ']'
+ '[' --build == --build ']'
+ create_clean_directory build
+ dir_name=build
+ '[' -d build ']'
+ '[' -f build ']'
+ mkdir build
+ echo 'Clean directory created: build'
Clean directory created: build
+ return 0
+ qmake -unset QMAKEFEATURES
+ git submodule update --init --recursive
Submodule 'gui/app/third_party/qnanopainter' (https://github.com/QUItCoding/qnanopainter) registered for path 'src/third_party/qnanopainter'
Cloning into '/home/pi/VentilatorSoftware/gui/src/third_party/qnanopainter'...
Submodule path 'src/third_party/qnanopainter': checked out '8a74ead156ec81f27035c0f60d509cb93df6d2ce'
+ cd build
+ qmake ..
Info: creating stash file /home/pi/VentilatorSoftware/gui/build/.qmake.stash
+ make -j
cd src/ && ( test -e Makefile || /usr/lib/qt5/bin/qmake -o Makefile /home/pi/VentilatorSoftware/gui/src/src.pro ) && make -f Makefile 
Project MESSAGE: QNanoPainter Release build
Project MESSAGE: Including QNanoPainter OpenGL ES backends
Project ERROR: Unknown module(s) in QT: serialport
make: *** [Makefile:47: sub-src-make_first-ordered] Error 3
+ '[' --build == --test ']'
```

martukas, 2020-06-17:
Installing `ibqt5serialport5` didn't help.
Subsequently installing `ibqt5serialport5-dev` finally helped.
One and/or both of these fix the problem.
Should be documented.

martukas, 2020-06-17:
Ok, I now see that dependency installation is something automated in the `gui.sh` script. The docs should reflect this.

Also, I had a "virtual memory exhausted" hangup while building with the script.
Going into `gui/build` and running `make` manually without the `-j` option finally got me through it.
This is somewhat brittle.

paulovap, 2020-06-18:
Pi has limited memory, so having jobs in parallel might exhaust it.

martukas, 2020-06-19:
Actually, I think the documentation could be even better, and the `-j` thing is still sort of outstanding. I will take care of this later.

martukas, 2020-06-20:
should be run with `--serial-port /dev/ttyS0`

PR #556 (2020-06-18): [GUI] Remove QtCharts dependency and update Readme
Fix #543
closed 2020-06-25 by martukas