RespiraWorks/Ventilator #552: GUI should log to disk

martukas, 2020-06-18:
**What has to be done**
GUI app should write logs to disk always. This should at the least include any serial communications failures, timeouts, and possibly some debug information. For now it should be something sufficient to help diagnose any problems in getting  physical setups to work.
Describe what has to be done, ideally in declarative form - what does success look like.

**How do you know it has to be done**
We had issues with accessing the correct tty device to communicate with STM32. Without console, it would have been hard to diagnose the problem. Since we intend to deploy and run GUI without X11 or recourse to console for the user, the necessary information should end up saved somewhere that can be checked and reviewed via SSH at a later time.
I believe requirements also require alarms logging to be done. That's the bare minimum. I think we may want to log much more detailed stuff, especially while still in development.

**Code pointers**
There are some instances where error messages are written to `qCritical()` or `qFatal()` or similar. There is probably a way to configure the Qt logger to have a sink for writing to files. Possibly to system logs? Please also document where the files end up and how to easily access them. Please also write some tests to confirm this functionality.

martukas, 2020-06-26:
@paulovap @a-vinod maybe y'all had some insight on this?

paulovap, 2020-06-26:
Yes. `qCritical()` and friends can be redirected to a file. Where should this lie in the filesystem?

closed 2020-08-14 by martukas