RespiraWorks/Ventilator #554: Script test analysis/evaluation

martukas, 2020-06-18:
**What has to be done**
Some visual evaluation of test data could be replaced and/or complemented with scripted analysis. We have talked of things like:
* undershoot
* overshoot
* not falling down to PEEP
* bad/good slopes or curves for reaching PIP
* evidence for lung compliance by inspection
* oscillation
In addition to this, we have been able to say if a particular test was performed with satisfaction or not. 
It seems that at least some of these evaluations could be abstracted away with python scripts that analyze collected debug data in post-processing. A report could be generated that identifies sore spots.

**How do you know it has to be done**
More people have been testing the controller with their pizzas and hand-made lungs. Not everyone is equally adept at understanding the implications of the test data. Since the home made test lungs perform differently, it is important to tally up the ways in which they are different so we know what efforts we can make to conform them to a standard. Also, we will want to automate this much more of these tests in the future, and evaluating the data automatically will become important.

**Code pointers**
The `/utils` directory has a script for running covent tests. Test data analysis could possibly live therein as well.

tapemaster, 2020-07-22:
Would be good for Data Scientist to pick up.

martukas, 2020-07-24:
As a minimum, we need the existing performance test scripts to generate a file with metadata, and help us avoid very long directory names that we now have in `sample_data`.

tapemaster, 2020-07-29:
Can we benefit from asking VentMon about this?

martukas, 2021-04-07:
This is a signal processing or data science task for a python-minded person.