RespiraWorks/Ventilator #591: BTPS Integration

esc-works, 2020-06-21:
Leaving this here to address later—sorry if I am bad at writing issues. 

BTPS Correction is used to convert flow and volume measured at ambient conditions to the conditions within the lungs. Ambient conditions are called ATP (ambient temperature, pressure); the conditions within the lungs are called BTPS (body temperature, pressure, water vapor saturated).

Currently, the flow measurement and integration does not correct for the fact that the inspiratory flow is generally at atmospheric temperature, pressure, and humidity, while the expiratory flow has been saturated with moisture in the patient as well as increased in temperature. This has not been an issue during prototyping as there is no heat or mass transfer in the test lung. 

The following link describes a standard correction to account for this fact.

https://d3cumwfln92soa.cloudfront.net/8a716ac3fd123ce0becd7b56596582d4fc4c0c47/appnote-btps-correction-v01r.pdf

I believe by my review this can be implemented solely in software without additional instrumentation, but only if the user enters approximate values for altitude and current temperature. We may later want to add sensors, I may make another issue on the hardware side here, as I believe this can be relatively easily (at least, pressure and humidity) accounted for in the circuit board with an additional sensor.

neelfirst, 2020-06-21:
Related HW issue: https://github.com/RespiraWorks/SystemDesign/issues/17

jlebar, 2020-06-21:
> sorry if I am bad at writing issues

This is a fantastic issue.  Thanks, Ethan.

One thing to note, volume zeroing will "fix" this.  The problem is, volume will be off by 10% (if I'm reading this document correctly), and the shape of the volume curve also won't look quite right.

martukas, 2021-01-18:
Related to #652

martukas, 2021-04-07:
We have to way to confirm that our corrections are correct without having humidifier, heater and probably more sophisticated test equipment.