RespiraWorks/Ventilator #611: Robust overpressure protection

esc-works, 2020-06-25:
Numerous options have been tested. All of these present failure modes, or had leakage issues which were likely more significant than their pressure protection function. The current design does not possess an additional overpressure protection relief valve. The fan specification is such that it likely cannot result in a pressure larger than what would be specified by a relief valve (~60 cm H2O). However, the PSOL can. There is some leakage at high pressure through the pinch valve, but this is likely not sufficient. This is an area requiring further address.

inceptionev, 2020-06-25:
Task: find assignee to help with this now that Ethan is way too employed

martukas, 2020-07-23:
@inceptionev do you think this is actually a hardware issue?

inceptionev, 2020-07-30:
@martukas this is a hardware issue, yes.

This ticket may be a duplicate of #656

inceptionev, 2020-08-07:
India team has been working on this and selected this valve 002-019-1003: https://respiraworks.slack.com/archives/C0111E2GFPF/p1596284702108100

inceptionev, 2020-08-30:
This has been superseded by #656

closed 2020-08-30 by inceptionev