RespiraWorks/Ventilator #613: OXY: determine control values for VentilatorSoftware for v0.2 (CoVent demo hardware)

inceptionev, 2020-06-25:
same as RespiraWorks/SystemDesign#136, but for the oxygen PSOL controller

- determine exhale valve PEEP setting (maximum position for achieving highest expected PEEP with inhale valve fully open)
- determine exhale valve PIP setting (minimum position for achieving lowest expected PIP with inhale valve fully closed))
- determine acceptable kp value
- determine acceptable ki value
- run test case and post data

note that these values are being selected for highest performance, not lowest gas consumption. Once these values have been established, they can be tuned for other cost functions.

martukas, 2020-06-25:
@inceptionev do we consider this done?

inceptionev, 2020-06-25:
Done!

inceptionev, 2020-06-25:
closing issue, completed by b514f94b0de66710a43f8e3bb138cbb976c54fca

closed 2020-06-25 by inceptionev