RespiraWorks/Ventilator #629: Blower and Pinch Valve Life Leader Test

inceptionev, 2020-06-25:
ticket to track the progress of the life leader test.  The life leader test was started May 28th and is running at full throttle with the pinch valve running from full 0 to 100% at an RR of 60 (one cycle per second).  The test will run roughly 24hr/day with short shutdowns for inspection and when running tests or filming on another set of hardware where the noise is interfering.

inceptionev, 2020-08-30:
test has been performed to 1108 hrs, no obvious mechanical failures observed, pending final performance comparison to a new valve.

inceptionev, 2020-08-30:
This is roughly 4 million cycles, representing about 12 continuous weeks of operation.

martukas, 2021-01-18:
Life leader was wrecked while shipping by USPS. I received it broken on Sept 6, 2020. Shipped back to @inceptionev, further status pending.