RespiraWorks/Ventilator #631: Venturi dP characterization coefficient

esc-works, 2020-06-25:
One issue with the reliability of the venturis is that we are using an "ideal" response coefficient, but we are working in the physical work. @sglow has done some calibration with an air cylinder, and found the response was reliably 12% low, 
[image]

We have done some testing with the pneumotachograph, and looks like the flow was about 12% high. 
[image]

Both look highly linear though, so I think we need to do better about coming up with a characterization curve since it probably has to do with manufacturing variation and consistency in how we are doing the calculation.  

1. measure the throat and channel diameters with a micrometer and pin guage to remove any variation in the calculation of throat diameters. 
2. Calibrate a known diameter venturi against the pneumotachograph and develop a linear offset correction factor.
3. Try printing an additional venturi with slightly different throat diameter to check scalability of correction factor.

inceptionev, 2020-08-30:
making a comment here to link this ticket to the archived slack channel #calibration-tests where this some of this discussion took place: https://respiraworks.slack.com/archives/C014BB987TQ

martukas, 2021-05-05:
Venturi geometry has changed and we probably need to redo this. Furthermore, we probably want to come up with a reproducible protocol that can be easily learned and used by people not intimately familiar with this. This would go with using the Ventmon.