RespiraWorks/Ventilator #646: Oxygen inlet fitting, check valve, filtration specification

ebhillstrom, 2020-06-25:
1. There's a possibility of unwanted junk (coarse debris, maybe stuff that flakes off in the piping) entering on the O2 input. One possibility is to use a metal screen filter for O2 input. A possible vendor:

https://catalog.okeefecontrols.com/category/miniature-in-line-screens---breathers

2. Find and implement the standard for oxygen inlet fitting for front of device.

3. Need check valve downstream of oxygen inlet?

martukas, 2022-01-22:
Related to #714

martukas, 2022-01-22:
Check valve question tackled in separate ticket #1197

inceptionev, 2022-07-12:
We will probably want to use DISS-compliant medical oxygen fitting such as this one: https://mfimedical.com/products/allied-healthcare-diss-male-to-1-8-npt-male-fitting-with-check

inceptionev, 2022-07-19:
We can use something like this to mount the DISS fitting to the acrylic: https://www.mcmaster.com/7880T341-52065K401/

martukas, 2022-07-19:
Some sort of fittings from #996 would go on the inside end of the acrylic mounted inlet.

inceptionev, 2022-08-07:
The 1/8 NPT DISS fitting will not fit the specified (1/4 NPT) panel mounting solution.  Here is an alternative:
- 1/8 NPT DISS fitting: https://mfimedical.com/products/allied-healthcare-diss-male-to-1-8-npt-male-fitting-with-check
- 1/8 NPT Panel mount: https://www.mcmaster.com/50785K272/
- 1/8 NPT to 8mm Push-to-connect tubing fitting