RespiraWorks/Ventilator #652: Humidity / Temperature / Density Variation in Measured Flow

esc-works, 2020-06-25:
Given the current baseline (see RespiraWorks/SystemDesign#9) flow measurement solution (a venturi) there is a dependance on the density of the gas, which changes with temperature and humidity. If there is no patient, then there is no variation. But with downstream patient connections, the humidity and density could change.
So the next steps would be
-Check how much the humidity could affect the density, and how much that changes the measured results. What's the worst case uncertainty.
-If needed, research options (BME280 breakout board for example) for P/T/RH sensor and integrated housing.

inceptionev, 2020-08-30:
This is a related ticket for fio2 calibration (i.e. is 50% fio2 50% of the hospital supplied oxygen concentration?): #859 

The ambient pressure / temperature sensor is being added to the next revision of the PCB: #814

As well as to the Rev 1 Interim PCB, created as a stopgap dev tool to allow software development on Rev A features using the Rev 1 PCB: #884

inceptionev, 2020-08-30:
Note that is both a technical and a requirements discussion

martukas, 2021-01-18:
This is pending integration of new electronics, therefore rescoped to 0.4