RespiraWorks/Ventilator #654: Select oxygen control solenoid

ebhillstrom, 2020-06-25:
Not currently implemented in alpha. 

Requirements: 
- Normally Closed, Cv ~0.1-.2
- 50 psi
- ¼” NPT
- O2 cleaned/compliant

The O2 control valve is a binary (on/off) solenoid that gets a timing signal from the vent controller and is inside the vent. The O2 regulator is totally separate. It's possible we can get away without it.

Three vendors we've tried:

Burkhart - pretty helpful, Colin Radmakker. Has sent us free valves.
Reinhold automation (MAC valves) - kind of helpful, no free valves, think they misunderstood the spec
US Solid - Quality/lifetime insufficient. Don't think we can put these in production hardware.

martukas, 2020-07-23:
https://www.smcpneumatics.com/PVQ31-6G-40-01N.html

VSTheBestDeveloper, 2021-01-16:
This ticket was Closed in the Hardware project. Moved it to Mechanical and Pneumatics and closed it.