RespiraWorks/Ventilator #655: Proportional valve design tracking.

ebhillstrom, 2020-06-25:
(Editing The original post to have more background - Ethan)
Originally, the flow loop was simple: 
[image]
This worked great in terms of having a simple system. The loop worked by using reserve flow rate capacity to generate backpressure against the tuning valve. This worked, but during inhale, the majority of the flow went out the exhaust. This became an issue when we started to look at Oxygen consumption. The problem is we need to flow oxygen to the patient, at least 30-40% by volume up to 100%. But oxygen isn't a freely available resource, and for patients on high FiO2, we could easily exceed the flow capacity of patient O2 supplies trying to supply high flow rates while wasting most of it. So we decided to include a high-flow solenoid on the exhaust, with a Cv ~4.8 ([flow rate of gallons/minute against 1 psi](http://www.idealvalve.com/pdf/Flow-Calculation-for-Gases.pdf)). This works and is in the current model. The problem is we were using a US Solid solenoid. Asking them repeatedly, they only guarantee 500,000 cycles (a little over 1 day in constant use). We sourced a solenoid from Burkhart for 100,000,000 cycles and its $220, and for the most part seems way overkill for the device.
This is a high flow valve. Specification is as follows:
- Pressure drop: 60 lpm through the valve with a pressure drop of less than 0.49 kpa (nominally a Cv of 2.3).
- Normally open is strongly preferable.
- Pressure is never more than 1.5 psi on either side.
- It needs to open at 0psig (direct lift).
- It doesn't need oxygen cleaning.
- 12V DC

We'd consider two valves in parallel if that was required to get the flow/pressure drop with something in stock.

Three vendors we've tried:

1. Burkhart - pretty helpful. Has sent us free valves. but they are $220 dollars
2. Reinhold automation (MAC valves) - kind of helpful, no free valves, think they misunderstood the spec. 
3. US Solid - Quality/lifetime insufficient. Don't think we can put these in production hardware.

neelfirst, 2020-07-15:
More detailed discussion of pinch valve and anti-asphyxiation brought over from slack (copied and condensed from #658)

- stepper-driven pinch valves are the primary source of pressure control and snap open in the absence of power.
- In a power failure case, both the inspiratory and expiratory valves will open, through which the patient will be able to breathe.
- We will not know until the full loop is built how much resistance there will be.
- Due to this, and the simple design of the pinch valves, it is highly unlikely that a mechanical failure will result either of these valves becoming stuck-closed. The likelihood of both becoming stuck-closed is extremely small.
- This leaves the case in which the valves are actively commanded by the electronics to stay closed. In the event of a computing failure in either the UI Computer or the Cycle Controller, a watchdog timer monitors each and will reset the offending computer, which should then be able to regain control of the valves.
- Furthermore, it will need to be confirmed in the final design that the UI Computer cannot command any modes that will cause the Cycle Controller to hold these valves closed.
- Assuming this is the case, then only the cycle controller needs to be considered in the anti-asphyxiation concept of operations.
- In that case, the Cycle Controller watchdog timer can also be connected to the stepper motor drivers which will default them to an unpowered state in the event of a Cycle Controller computing failure, which will again cause them to snap open, adding another layer of anti-asphyxiation protection

inceptionev, 2020-07-30:
We have concluded to use the stepper-driven pinch valve for blower control and a proportional solenoid (PSOL) for oxygen control.  

TODO to close this ticket:
- Confirm that these updates are reflected in git. (true for pinch valve, confirm for PSOL)

martukas, 2021-01-18:
Testing with this valve has so far been successful. Specific design of mechanical components is superseded by #941

closed 2021-01-18 by martukas