RespiraWorks/Ventilator #656: Selection of overpressure relief valve

neelfirst, 2020-06-25:
On patient inhale side, after O2 mixing, before humidification.

TBD: before or after inhale pressure/flow measurement?

Rafee-UM, 2020-08-03:
From INDIA side we have ordered an over pressure relief valve as per the attached data sheet. Once we got that from the vendor we test it and will update the results. **002-019-1003** is the model which we ordered. 
[SAFETY RELIEF VALVE CATALOGUE.pdf](https://github.com/RespiraWorks/Ventilator/files/5015467/SAFETY.RELIEF.VALVE.CATALOGUE.pdf)

Rafee-UM, 2020-08-03:
As discussed here https://respiraworks.slack.com/archives/C0137BLF17A/p1590440382159500, INDIA team is planning to develop a safety relief valve with help of Umbrella type check valves, which will have absolutely zero leak until the crack pressure is arraived. These are made of medical oxygen compatible elastomer material.
      we are getting the sample quantity of Umbrella valves, once we recieved them, we will revert back with test results. please find the attached umbrella type check valve as your reference. [UM 070.006 SD.PDF](https://github.com/RespiraWorks/Ventilator/files/5015521/UM.070.006.SD.PDF); 
[UM070.006-151.01.pdf](https://github.com/RespiraWorks/Ventilator/files/5015524/UM070.006-151.01.pdf)

inceptionev, 2020-08-30:
Note 2x of Generant VRV-750B-V-0.8-X have been ordered for the US build.  Additionally, two equivalent GRVS units have been ordered for evaluation.  Expected Oct 11th.

Next steps:
- Complete the US enclosed ventilator build
- Compare notes with India build
- Do a complete design review

inceptionev, 2020-08-30:
This comment copied over from duplicate ticket #611 (now closed):
India team has been working on this and selected this valve 002-019-1003: https://respiraworks.slack.com/archives/C0111E2GFPF/p1596284702108100

martukas, 2020-11-04:
Turn out that what we have ordered is actually VRV-250B-V-0.8, which can be decoded using this:
https://www.generant.com/wp-content/uploads/2019/09/SA.SL_.VRV001.G.4514-Series-VRV-Product-Literature.pdf

martukas, 2021-01-06:
The valves have arrived. I think this can be closed for now unless testing reveals the need for an alternative.

martukas, 2021-01-15:
In discussion with @tijanim a problem has been identified that might possibly be a fire hazard. The overpressure valve comes after oxygen mixing and releases the gas into the enclosed ventilator assembly, where electronics reside in the open. In case of 100% oxygen, there may be fire hazards. If this is true, we need to select an "inline" version of the valve, so that we can direct the output to be dumped outside of the ventilator unit.

@jmriphagen @inceptionev @mrkensan  thoughts on this?

inceptionev, 2021-01-15:
Good catch!  Would it make sense to route the patient outlet such that that
the relief valve can be placed outside the unit?  Is there a easily
sourceable inline version of this valve?

inceptionev, 2021-01-15:
Ok, looks like the inline version is just VRVIxxx (with the same parameters)

martukas, 2021-01-28:
Related question -- what pressure should tubing be rated for the relief valve outlet?