RespiraWorks/Ventilator #657: Finalize filter and filter housing

neelfirst, 2020-06-25:
- what kind of filters are we using (alpha: roomba hepa filters)
- does the filter housing fit the filters
- can the filter housing be mounted on the enclosure

DonnaCG, 2020-07-30:
Design to be reviewed with enclosure build

inceptionev, 2020-08-01:
a BOM and assembly page has been started: https://github.com/RespiraWorks/Ventilator/tree/inceptionev/add_filter_instructions/manufacturing/alpha-build-instructions/filter-holder

DonnaCG, 2020-08-01:
overall BOM has been updated to reflect current design (In GDrive): 
https://drive.google.com/file/d/1i01Zh0hXRvrTphOLACNOmgj-qOsmSzOZ/view?usp=sharing

martukas, 2020-11-04:
Actual design content should not be stored in tickets but committed to the repository properly.

Recent builds have revealed many problems of the "current" filter housing incompatibility with enclosure, lack of discipline in versioning when using OnShape and swath of general hard-to-track-down problems. We have also decided to migrate all CAD to AutoDesk Inventor.

This ticket is superseded #917, and therefore is being closed.

closed 2020-11-04 by martukas