RespiraWorks/Ventilator #658: Blower response time is too slow

neelfirst, 2020-06-25:
John Batteh: We definitely need to nail down the blower inertia value.  Currently it is set at 0.69s but with this sort of time constant, I think the response will be pretty slow.  We'll set up the latest test to try and achieve 20 breaths/min and see how the system responds.

Just thinking about it quickly- there will definitely be a problem and we don't even need to run a sim.  For 20 bpm and IE=1/2, then the intake event is 1s long.  So if the time constant is 0.69s, your speed response alone will make it difficult to hit your PIP target.  On your target hardware, are you able to run something like the breathing parameters above and hit the PIP target?  If so, that is interesting and we need to re-evaluate the blower inertia.  It is critical.

inceptionev, 2020-07-15:
@neelfirst with the pinch valve implemented, is this ticket ready to close?

neelfirst, 2020-07-15:
I know in practice the pinch valve is done and is our primary solution for improving blower response time. So this ticket can be closed.

However I don't see enough of a paper trail in GH issues that declare the pinch valve fully implemented. In particular - 
#655 should have at least a pointer to the final design and
#629 should have a pointer to the completed test data, for that design.

The Slack discussion for the pinch valve failure modes should be in #655 and tagged for future FMEA work. (did this)

inceptionev, 2020-07-30:
Decision is generally made, pending completeness of test data.  Closure pending closure of the two tickets called out above.

martukas, 2021-01-18:
Life leader pinch valve has proven valve to work for over 4 million cycles (12 continuous weeks of operation). Tracking of testing continues in #629
Mechanical design of pinch valve superseded by #941
We consider the general question closed, and specific response scenarios will be addressed in detailed integration testing with a more mature design.

closed 2021-01-18 by martukas