RespiraWorks/Ventilator #659: Oxygen Mixing Solution

neelfirst, 2020-06-25:
One of the requirements in the system is to provide both 50% and 90%+ oxygen (FiO2). See the [RMVS Gas & Electricity 3a](https://assets.publishing.service.gov.uk/government/uploads/system/uploads/attachment_data/file/879382/RMVS001_v4.pdf) for details. This was originally thought to occur upstream of the fan, before fire risks were identified flowing high concentration oxygen through the fan. It was decided to try and integrate the O2 mixing downstream. Oxygen is expected to come from wall supply or tanks. There is at least going to be a pressure regulator if not a rotameter to display flow. The baseline design is shown below:
[image]
Here, oxygen is constantly mixed into the system on both inhale and exhale. The flow is constant set by manually (User Action) tuning the regulator flow rate. The device should provide an indication of FiO2 as feedback. 

Another tweak is to use an O2 control solenoid to switch the O2 supply on only when it is during an inhale cycle. This potentially allows higher O2 rates, limits the risk of O2 backflow into the fan and limits the risk of inadvertently over-pressuring the patient during inhale (if the O2 flow valve is too high). 
Below is a subsection schematic:
[image]

inceptionev, 2020-08-02:
Work on this is being summarized here: https://docs.google.com/presentation/d/1tEIRDNIFct4DQpMUWZZn-SIk56GaNQ7PR2nPpMT_X8Y/edit#slide=id.g8ee06a2ec6_0_0

inceptionev, 2020-08-07:
See above document.  A working proof of concept based around flow control has been demonstrated.  Some pneumatic elements will have to change, and a sensor will need to be added.