RespiraWorks/Ventilator #706: evaluate need for RH sensor in humidifier

neelfirst, 2020-06-25:
Can we get away without one and just do open loop control on the humidifier?

or do we need to target a specific RH?

What is the tolerance on the RH?

Possible sensor: https://www.sparkfun.com/products/9569

VSTheBestDeveloper, 2021-01-16:
This ticket was Closed in the Hardware project. Moved it to Mechanical and Pneumatics and closed it.

martukas, 2021-01-17:
It appears that this was not superseded by another ticket and we still lack formal evaluation of this question.

zombiewater, 2022-07-07:
Given that humidity changes throughout the day, depending on the weather and the environment they are in, we would want the patient to be comfortable with the air they breathe in. I think we do need a controller. I believe tolerances would be need to be in within 1C of the corresponding RH change based on the research I just did. 
[honeywell-sensing-sensors-ventilators-application-note-009041-11-en.pdf](https://github.com/RespiraWorks/Ventilator/files/9059527/honeywell-sensing-sensors-ventilators-application-note-009041-11-en.pdf)
[WHITE-PAPER-Relative-Humidity-Sensor-Behavior-and-Care.pdf](https://github.com/RespiraWorks/Ventilator/files/9059528/WHITE-PAPER-Relative-Humidity-Sensor-Behavior-and-Care.pdf)