RespiraWorks/Ventilator #712: Support persistence storage

tapemaster, 2020-06-25:
We need to store the latest ventilator settings on the storage which survives power disconnect.
Options:
1. Doing it on the flash of STM32 would make the processor stop for a short time because the program memory and data memory are on the same flash bank.
2. Get another version of STM32 with dual flash banks.
3. Add an EEPROM on the PCB, this will not pause the STM32 when it needs to write/read something there.

Currently the preferable option is number 3.

jlebar, 2020-06-25:
Specifically, we're ok with (1) if and only if we don't have to write anything while the vent is hooked up to a patient.

jlebar, 2020-06-25:
Related: https://github.com/RespiraWorks/VentilatorSoftware/issues/376
https://github.com/RespiraWorks/VentilatorSoftware/issues/170

tapemaster, 2020-07-22:
@inceptionev can you please known details about EEPROM here?

inceptionev, 2020-07-22:
This is the current selected i2c EEPROM:https://ww1.microchip.com/downloads/en/DeviceDoc/24AA256-24LC256-24FC256-Data-Sheet-20001203W.pdf

PR #995 (2021-02-10): Jtravert/issue 712 add eeprom
Rewrite from #936 (which died due to #980) taking @martukas' review comments into account except:
- i2c1 is still a global variable, will make it disappear in a future commit 
- concern about void* in I2C::Request, I2Ceeprom::WriteBytes|ReadBytes and NVParams::Handler::Set|Get, which I see no (clean) way around
- NVParams local functions are still local
- concern about needing additional protection for NVParams reset
- ~~mock is still done with localized #ifdef~~ e4f5c5c is hopefully solving this the way you expected

Still needs some additional work on unit tests [... cut]
closed 2021-02-10 by asmodai27