RespiraWorks/Ventilator #713: Select New Blower

inceptionev, 2020-06-25:
New blower must:

- Support full spec PEEP pressures in Requirements at spec flow rates
- Support high-altitude operation per Requirements
- Have enough control margin for good control margin

esc-works, 2020-06-29:
Hey Edwin, can you expand on the second bullet point? Do you think high-altitude requirements would drive us out of spec on flow pressure? We should also have a conversation about whether we need to change the design to meet a spec or if the spec is just something we won't meet; I think 40cm seems defensible if we can get there; 

if we're trying to get to 60m and that we need new fans; maybe just if you have a second to expand here where (spec-wise) we're short with the current fans.

inceptionev, 2020-06-29:
Sure.  The design currently is marginal even for 40cm.  The blower is capable of the pressure in static conditions, but drops off steeply with flow.  So in large tidal volume conditions, we may not be able to meet the PIP pressure within the inspiration time, for example in CoVent test #4, especially at an altitude of 10,000ft where the air density is 30% reduced (and we should expect a similar 30% reduction in pressure).

Much of the feedback that we have received in talking to doctors have indicated that PIP pressures of up to 60cmH2O are desired, so I believe that is where our RW spec comes from.

It's not out of the question that we could derate our specs to use the current blower, but we would need to do some additional testing to verify this since the margin is so small, and also the spec would probably need to have some conditionals attached to it, such as, can support PIP of 40cm at up to 10,000ft, but only if TV is < [some number].

Having more margin in the blower (and consequently, the PSOL as well, see issue #714) would give us more confidence in this operating envelope.

martukas, 2021-01-18:
This is related to #925 pending further testing with the alternative types mentioned in that ticket.