RespiraWorks/Ventilator #714: Select New Oxygen Valve

inceptionev, 2020-06-25:
New Oxygen Valve must:

- Be Oxygen safe
- Support highest PIP requirements at max flow rate
- Have good control margin

esc-works, 2020-06-29:
I assume this is the oxygen control valve? I think the big thing we were short on here was max flow rate design. As discussed with @martukas and @sosaysthecaptain we wanted to try a pinch valve style oxygen regulator.

One suggestion, in addition to moving to much smaller tubing, was to change the current pinch valve to improve oxygen response was to add an arced, rather than flat pinch plate, giving a bit more angular resolution to the orifice size.

The other option is to find a higher CV proportional solenoid.

inceptionev, 2020-08-07:
The SMC PVQ31-6G-40-01 proved to be capable of the 60 lpm required to pass the CoVent tests.  The data for that test is here: https://github.com/RespiraWorks/Ventilator/tree/master/software/sample-data/2020-06-21-Peak-Oxygen-Flow

We are now also testing the UFlow IBV19M for the Indian market.  That test data is here: https://github.com/RespiraWorks/Ventilator/tree/master/software/sample-data/2020-07-31-UFlow-Performance

martukas, 2021-01-27:
Related ticket #922