RespiraWorks/Ventilator #721: Get webcam reliably stream from hardware test server

martukas, 2020-06-26:
**What has to be done**
There are two webcams connected to  "pizza farm" server, that will allow devs to remotely observe prototypes in action. They have been confirmed to work locally, and even managed to stream to web in a few ways. We need them both to be able to multicast (to be watched by more than one person at one IP) with video and sound, either on demand or constantly.

**How do you know it has to be done**
We need this for devs that do not have access to hardware. Both visual and audio feedback are useful when experimenting with a unit manually. We want it multicast so people collaborating or discussing behavior can observe the same system simultaneously.
We can also use this to promote and publicize our development.

**Code pointers**
It has worked with `motion`, multicasting, but without sound.
It has worked with sound but only streaming to a particular IP.
There is some notes and code under software/controller/integration_tests/webcam.
Please ask @martukas , @twslankard or @rwilbur  for server credentials.

martukas, 2020-10-15:
@danyal-turkoglu had been working on this in the following branch:
https://github.com/RespiraWorks/Ventilator/tree/issue_721_webcam_livestream

martukas, 2020-10-15:
To set up a Live Stream in Youtube, you need to 
* Click on RespiraWorks logo (top right)
* Youtube studio
* Create (top right)
* Go live

martukas, 2020-10-15:
What we want as a result is:
* Live channels with persistent URLs, but unlisted
* A script that will either start streaming them on demand or shut down the streaming (to conserve bandwidth)
* It should be clearly documented somewhere how this is done so people wanting to test don't need to read too much technical stuff and learn the whole toolchain that makes this work.

martukas, 2020-10-15:
The devices should somehow be addressed in a way that works after a reboot. Not sure if any of the audio ir video addresses change. The two webcams are identical.

pabbathishyam, 2020-10-23:
Hi @danyal-turkoglu - Thanks for creating the script in issue_721_webcam_livestream branch to send the live stream on YouTube. I used the script and able to stream the webcam to the live on YouTube.

The stream-to-youtube.sh script worked with following changes. Please update the script in your brnach:
1. Line# 16, KEY= "$1", please remove the space after KEY.
2. Line# 25, Remove the " after $KEY.
3. Add "fi" after the last line, to complete the if block.

pabbathishyam, 2020-11-06:
Please see the README info of webcam and relay8 for info on how to how to turn the light and start live streaming from the YouTube (Lights/Camera/Action).

martukas, 2022-06-28:
Completed in #914

martukas, 2022-08-13:
Server is back online from Oakland and these scripts need updating. As they currently are in the repo, it is not totally clear
* how the 2 scripts under /usr/local/bin are different from each other
* how they are different from what is committed to repo
* no script to deploy what is in the repo
* no good instructions or scripts for discovering all video devices in case something changes in configuration

Basically, it's time to update this, make it a bit more user-friendly and future-proof.

martukas, 2022-10-08:
This might be useful:
https://github.com/mpromonet/v4l2rtspserver