RespiraWorks/Ventilator #725: Develop integration tests for pinch valve

martukas, 2020-06-26:
**What has to be done**
Figure out ways to test the following for the pinch valve:
* test that it's wired correctly and rotates in the correct direction
* test homing of pinch valve
* automatic linearization / calibration or confirmation of linearlization
Tests should be fully automated and without human intervention. They should answer in the affirmative or negative - is the subsystem working correctly? Alternatively, as for the calibration, for example, it should record some values that can be compared for deviation from expected values.

**How do you know it has to be done**
Right now we are making a lot of assumptions about correctness of the lower layers of software and electronics. We should be testing some of those layers on a regular basis, which should also help us establish troubleshooting and regression procedures for QA and validation later on. Some of these tests might be adaptable as self-tests in the final production units.

**Code pointers**
We have some alternative pinch valve designs that can incorporate endstops and/or rotary encoder. We will soon have other sensors external to the prototype itself that can be use to help with replicating characterization and linearization tests.
Ask @martukas about server credentials and connecting up a prototype for experimentation.
Ask @sglow about homing and linearization, as he has experience with those features.

martukas, 2020-07-23:
The Posifa flow sensor is hooked up to Pizza4 on the test server, which can act as redundant sensor for this type of test.

inceptionev, 2020-07-23:
Lineariation table is implemented here: https://github.com/RespiraWorks/Ventilator/blob/master/software/controller/lib/core/pinch_valve.cpp

asmodai27, 2021-04-28:
This resembles the kind of tests we need for #307, except we might want to integrate an external sensor rather than rely fully on the vent's sensors.