RespiraWorks/Ventilator #728: Debian package of GUI app

martukas, 2020-06-26:
**What has to be done**
There should be a debian package of our GUI app, so it can be updated using the standard `sudo apt update && sudo apt upgrade` mantra. The package should also install all apt-gettable dependencies if needed.

**How do you know it has to be done**
Currently, docker pull takes about 10 minutes to complete, and is multiple 100s of MBs. The GUI app is in reality much smaller and this much effort and time is not reasonable, especially with our rapid development.  It has been proposed that we could have a debian package.

**Code pointers**
@paulovap has noted that this should be pretty easy.

paulovap, 2020-06-27:
Yes, making the metadata do generate the `.deb` the file is rather easy. The only thing I don't know is how to build the project on CI targeting arm. Maybe @jlebar can help with this

jlebar, 2020-06-27:
Do you know how to build locally targeting ARM?

The CI scripts are pretty straightforward, they just set up the circleci container and then execute the build.  See `.circleci/config.yml`.

martukas, 2021-02-03:
This probably has knock-on effect on #765 #88 #539 and possibly closes some of them as obsolete.
We probably want one custom image of Raspbian that we still need to script the creation of and store somewhere.
This image should come preconfigured to apt get the latest GUI app.