RespiraWorks/Ventilator #733: Bring 3x Stepper Modules onto the board

inceptionev, 2020-06-28:
We originally had this as an optional module that could be plugged into the Nucleo board on this setup:https://www.st.com/en/ecosystems/x-nucleo-ihm03a1.html

We now want to have this driver chip directly on the board.  The documentation for this module, including schematic are in the link above.  The modules are designed to stack up to 3-high, with some resistor placement options to allow all three to be addressed on a single SPI bus.  This 3-stack implementation is what we want to emulate on our board.

KennethLaws, 2020-08-08:
this issue should be connected to issue 733 steppers on pcb

martukas, 2020-08-31:
This issue looks like a duplicate of #773. Is has this been superseded by that ticket and can be closed?

KennethLaws, 2020-09-02:
I believe the answer is yes

inceptionev, 2021-02-25:
closed by pr #1033

closed 2021-02-25 by inceptionev