RespiraWorks/Ventilator #745: Motor needs 5V command but STM32 is 3.3V GPIO out

inceptionev, 2020-06-28:
will need a level shifter

dcstraney, 2021-01-19:
New level shifter implemented in #947 by copying non-inverting level translator from FoundryM design.

closed 2021-01-19 by dcstraney

inceptionev, 2021-02-25:
Rework instructions for adapting Rev 1.0 PCB:

- Obtain [sparkfun level translator board BOB-12009](https://www.sparkfun.com/products/12009)
- Solder the following connections on the bottom side of the board:
  - PCB P2.16 - BOB LV
  - PCB P3.31 - BOB LV2 
  - PCB J5.1 - BOB HV
  - PCB J5.2 - BOB HV2
  - PCB J5.3- BOB GND
  - Stake the wires to keep them from moving.
  - Attach the BOB to board with two-sided adhesive tape.
 [image]
 - Cut the BLOWER_CTRL trace on the top side of the board as shown:
[image]

[image]