RespiraWorks/Ventilator #751: Do not fuse motor

inceptionev, 2020-06-28:
Up for debate: do we fuse the motor?  In some safety critical systems, we do not fuse the most critical component, because it must run no matter what.  I.e. all other fuses exist to protect this component, and fusing it adds a failure point.

dcstraney, 2022-06-23:
Update: On Rev. A, the situation is...
Main power entry and battery have fuses, but 12V to blower doesn't have a fuse.  (Blower motor driver may have a fuse, but that's outside the scope of this.)
Stepper motor drivers also feed from 12V rail - they aren't currently fused but there's a PCB footprint where we could populate fuses to protect the 12V against stepper faults.  Shorts which happen downstream of the stepper drivers (in the steppers themselves or the wiring to them) will be handled by OCP in the L6470.
Other things which feed from VLink, upstream of the 12V power supply, are the 2x 5V power supplies - these aren't fused, but are current-limited (protects against shorts/over-current downstream of these power supplies, but not against failures of the power supplies themselves).

Overall there's two separate power supplies in between the blower and the main DC supply (whether that's from an AC-DC brick, or a battery) with their own complexities, current limits, and many points of failure - so I'm not convinced this discussion is relevant anymore in its current form.  Testing and designing for tolerance to power supply faults at a variety of points in the system is 100% worthwhile, but is its own separate, broader subject as part of a general fault tolerance investigation.

inceptionev, 2022-06-23:
Agreed, this was up for discussion with a previous design effort, but really needs to be part of a comprehensive system-level fault effects evaluation.

dcstraney, 2022-06-24:
Ok, will close this out that as a non-issue, pending a future in-depth fault tolerance investigation.

closed 2022-06-24 by dcstraney