RespiraWorks/Ventilator #766: Collect power consumption information

inceptionev, 2020-07-01:
Collect power consumption information in the following conditions:

Input power: 12V

 - Fully on (all actuators running)
 - All actuators off, but with GUI+Controller+Touchscreen running

inceptionev, 2020-07-01:
- Fully on (21% oxygen, air only): 3.7A (44.4W)

Note: we don't currently have a mode that runs both the oxygen PSOL and the blower at the same time, so this consumption has to be estimated:

- With full current on the PSOL as well +400mA @ 12V: 4.1A total (49.2W)
- All actuators disconnected, computing only, i.e. UI Computer, Controller, and Touchscreen on only: 0.77A (9.24W)

KennethLaws, 2020-07-02:
> LEARN FAST: This email originated outside of HERE.
> Please do not click on links or open attachments unless you recognize the sender and know the content is safe. Thank you.

This is great. Thanks!

inceptionev, 2020-07-15:
Task completed, closing ticket

closed 2020-07-15 by inceptionev