RespiraWorks/Ventilator #773: Place 3x Stepper Driver on PCB

KennethLaws, 2020-07-09:
**When to create Feature request or Task?**
Install to be on next revision of pcb

**What has to be done**
Stepper drivers are now on shield boards on the STM32.  Plan is to put them on the pcb

**How do you know it has to be done**
This is to transition from POC prototype toward a production ready prototype.  This is part of a multi task project.

**Code pointers**
Hardware, not code

**Other**
Should be linked to the next revision milestone

KennethLaws, 2020-07-09:
Stepper motor is 2 A per phase, this sounds about right for a small stepper to control a valve
Stepper driver l6470: 
https://www.st.com/content/st_com/en/products/motor-drivers/stepper-motor-drivers/l6470.html
stepper motor info:
https://www.omc-stepperonline.com/nema-17-bipolar-59ncm-84oz-in-2a-42x48mm-4-wires-w-1m-cable-and-connector-full-d-cut-shaft.html

Data sheet for l6470 stepper driver gives 5W max power, 7A peak, 3A rms.  Our power dissipation if we run the motor at maximum should be based on 2 A or 2/3*5=3.3 W.  If we use the HTSSOP28 pkg, 22 deg/W we should expect 3.3*22=72.6 deg rise.  Given the temp limit on junction of 150 deg C, we should be fine. 

Thermal relief or direct connect:
Thermal relief is recommended for parts that are through hole connected.  This is to avoid unreliable solder joints and mostly applies to hand soldering.  For reflow soldering and smd components the whole board is heated and so thermal relief is not needed
Interesting discussion here:
https://www.pcblibraries.com/forum/thermal-relief-for-smd-components_topic1009.html
Another interesting article here:
https://www.cysca.ca/pdf/an201211-01a.pdf

Connectors to motors:
Choosing to have one connector per device to reduce total current in the connector.  It was suggested to make the on board connectors different to avoid miss-connection but the steppers have identical connector included so we lose this unique connector advantage on the stepper end.  We also have low risk because if the motors are plugged in wrong there will be no damage since motors and wiring are identical. Miss-connection should be detected in a pre-use check that should be mandatory after any internal changes are made anyway.  Choose to use three identical connectors and recommend labeling motor cables.

Current capacity of vias:
This web page indicates:
1. 12mil via can safely carry around 1.2A of current, which is looser than the widely recognized 0.5 in the industry.
2. The larger 16mil, 20mil, and even 24mil via do not have obvious advantage in current carrying, and not linearly increasing.
Therefore, we recommend 10~12mil via to load current, higher efficiency, but also more convenient design. 

From <https://www.kingsunpcb.com/pcb-via-size-12mil-is-perfect-for-current/1707.html> 

Large Capacitors:
Choosing to have one 100 uF cap per driver because a single 330 uF cap would be further away from the Vs pins.

Driving the phases:
For conducting current from the capacitor to the chip, I want to minimize resistance.  Peak currents may be much higher than the 2 A rms per phase that the motor is rated for.  I have elected to put 10 vias on the capacitor positive pour and this feeds 2 vias on each of four VS pours connecting to the four VS pads on the chip.
The capacitor supply is connected by a 3.5 mm wide trace connecting to the fuse
12 vias on the supply trace on the top layer connect to the supply trace on layer 2

For conducting phase currents to the motor (2 A), circuit calcylator gives 0.781 mm trace width for external layers in air

Drill size:
Our small vias are 0.305 drill size is 12 mil, so we should be good in terms of avoiding very small drill sizes.

Power update:
15 W for steppers, 15/12 = 1.25 A
other: 60 W Blower + 10W UI computer + 5W Touchscreen + 2W Cycle Controller and sensors = 77 W (6.5 A)

From <https://app.slack.com/client/T0102KRCH5M/C011NJ8MHCJ/thread/C0100MWE4V8-1592602649.239100> 

total on 12 V receptacle = 7.75 A

Add 50% margin, gives 7.75 + 7.75/2 = 11.625 A (trace width: 8.85 mm in air)
4.425 mm top and bottom.  There is 3.96 mm between pins giving a max trace width of 3.96 mm on each of these pins to avid contact across the gap between pins.  50% margin is not possible.

Add 20% margin, gives 7.75 + 7.75*0.2 = 9.3 A   (trace width: 6.5 mm in air)

KennethLaws, 2020-07-09:
12 V in detail (in progress)
[image]

KennethLaws, 2020-07-09:
12 v detail bottom layer (in progress)
[image]

KennethLaws, 2020-07-11:
Bottom layer detail of 12V jack connection and one stepper layout (not complete)
[image]

KennethLaws, 2020-07-11:
Top layer detail of 12 V in Jack and one stepper layout
[image]

KennethLaws, 2020-07-24:
tented vias, put stepper components is defined room and modified schematic hierarchy, increased board size to enable clearance with Rpi but look like there still might be issues.  Might need to rotate the Rpi to enable clearance for usb connectors

KennethLaws, 2020-07-25:
100 uF cap going to ceramic, much smaller footprint, avoids collision with Rpi
[image]

KennethLaws, 2020-08-08:
Adding ports for controller signals to controller schematic
// PA5 - SCLK
// PA6 - MISO
// PA7 - MOSI
// PB6 - CS
layout for first stepper ~ done, added copy of schematic and duplicated room layout (lots of issues).  Attaching the above signals as part of process of clearing warnings and errors

KennethLaws, 2020-08-30:
[image]

Stepper schematic is not much changed other than to divide the 100 uF electrolytic capacitor into two 47 uF ceramics to save space

KennethLaws, 2020-08-30:
[image]
Board shape was increased in length by about 1" just make addition of the steppers easier, and to not change space for STM micro or the Rpi.  Not really intended as a final design, but with the understanding that, at least for now, leaving the positions for the headers was important for retaining capability to use standard shields.  The mounting holes should either be moved or additional holes added to give support to end of board to support stress when mating connectors.

KennethLaws, 2020-08-30:
[image]

3D layout shows no obvious red flags.  If board length is a big problem, a right angle connector might fit under the Rpi connectors and the stepper drivers pushed further under the Rpi outline.  For noise considerations it would be better to keep them further apart. As above.  There are some obvious errors in component names and name positions could be improved.

KennethLaws, 2020-08-30:
[image]
Top Layer
Disbursement of thermal vias could use some adjusting.  connection between this ground plane and the rest of the ground on the board is limited to the upper left corner.

KennethLaws, 2020-08-30:
[image]
Internal layer 1, ground plane.  the plane is uninterrupted except where it is insulated from signal carrying vias.  I just realized that I need to define a split plane here if I am to isolate this area from the rest of the board ground as I intended.

KennethLaws, 2020-08-30:
[image]
Internal plane 2.  Power plane for the stepper supply.  12V area is defined by outer red line and fed by the four vias at top rated at about 1.2 A each.  The red outline under U10 is a ground region to help evenly distribute heat conducted by the vias to the bottom layer

KennethLaws, 2020-08-30:
[image]
Bottom Layer, gap divides the stepper ground plane from the ground plane of the rest of the board

KennethLaws, 2020-08-30:
Just added a split plane to internal layer 1.  Not sure how the software will handle having a ground isolated from ground, all on the same net.  Assuming that the line defining the split plane has finite width and is not connected to any net.

martukas, 2020-08-31:
Does this ticket supersede #733 ?

inceptionev, 2021-04-26:
implemented by #1101

closed 2021-04-26 by inceptionev