RespiraWorks/Ventilator #776: Review relocation of Power & Oxygen

DonnaCG, 2020-07-17:
**What has to be done**
Review to see move power & Oxygen reconfig is required prior to enclosure fabrication.

**How do you know it has to be done**
Feasibility study is completed.

tijanim, 2020-07-17:
- Investigate moving O2 Hardware to underneath shelf
- Investigate moving O2 input port to side panel
- Add power port to side of panel.

DonnaCG, 2020-07-31:
Since these changes will not be included in Beta Build, we should move task to Reference Design or Gamma Build

martukas, 2021-01-19:
This is partly superseded by #965 (power equipment placement)
O2 related questions still stand

martukas, 2021-04-06:
Oxygen input through-fitting is now on the newly created back panel.
DIN power input is now on the newly created back panel.

This among the checked-in changes addressing #957 

[image]

closed 2021-04-06 by martukas