RespiraWorks/Ventilator #780: Investigate manifold to combine O2 sensor mount, venturi, and relief valve mount into one component

tijanim, 2020-07-17:
Figure out if we can simplify this arrangment into a single printed manifold that allows mounting of relief valve, O2 sensor mount, and venturi that fits into the existing footprint. 
[image]

inceptionev, 2020-07-17:
Oxygen sensor should be close to the flow for best response, and as far downstream from the oxygen inlet as possible to allow for mixing.

The path should probably be something like this: oxygen inlet > relief valve > venturi > oxygen sensor.

martukas, 2021-01-06:
* have a custom oxygen sensor adapter here: https://github.com/RespiraWorks/Ventilator/tree/master/manufacturing/oxygen#oxygen-sensor-sub-assembly
* This needs to be modified to include an adapter for the overpressure valve, which was selected in #656 
* It might also be a good idea to incorporate an additional elbow that redirects flow towards the front panel filter adapters, eliminating the need for one more McMaster elbow
* Integrating with the venturi is off the table because the venturis are upstream of the mixing chamber, and any further integration of manifolds is probably better left for after we have a definitive mixing chamber design (discussed in #915 )

martukas, 2021-01-18:
This has been partly addressed in in #916

PR #972 (2021-01-31): Relief valve manifold integration
PR for the addition of the 1/4-18 NPT hole onto the oxygen manifold for the releif valve

closes #780
closed 2021-01-31 by martukas