RespiraWorks/Ventilator #782: Build Ventilator - Equipment Panel - Location WEST

DonnaCG, 2020-07-17:
Assemble all components to Equipment Panel.

This is 2 of 2 Ventilator Beta Builds

martukas, 2021-01-18:
Sister ticket for MD build #781

inceptionev, 2021-02-25:
Build notes.

- Pinch valve hole patterns do not line up properly
- Pinch valve stepper motor cutouts needs enough gap for the wires to pass through during assembly.  Consider cutting a notch.
- Stepper wires needs a place to route up the the PCB
- Blower adapter does not retain gasket, makes it hard to keep it in place during blower stack assembly.
- Holes in the plates for the blower do not clear the screws, needed to be reamed.
- Mounting nuts from the back side for the oxygen regulator is pretty awkward, maybe consider using melt-in nuts?
- Check valve pressure drop seems pretty high.

inceptionev, 2021-02-25:
Build complete

closed 2021-02-25 by inceptionev