RespiraWorks/Ventilator #789: Build Ventilator - Display Equipment Panel - Location WEST

DonnaCG, 2020-07-18:
Assemble all components to Display Equipment Panel.
Document any issues/improvements identified during build.

This is 2 of 2 Ventilator Beta Builds

martukas, 2021-01-18:
Sister ticket for CA build: #788

martukas, 2021-02-05:
@inceptionev posted this in the mechanical channel:
"here’s how my PCB is mounted.  The screen has bosses for mounting an RPi.  I put standoffs into the bosses, then the PCB on top, with the long RPi standoffs now running through the PCB into the standoffs on the screen.  The the Pi just mounts on top of that."

[image]
[image]
[image]

inceptionev, 2021-02-25:
Build notes:

- Nucleo USB programming port is hard to access.

Build complete.

closed 2021-02-25 by inceptionev