RespiraWorks/Ventilator #800: Create page on oxygen filter

gracecanfield, 2020-07-18:
See details in https://respiraworks.slack.com/archives/C0137BLF17A/p1593033084325600?thread_ts=1592986029.321000&cid=C0137BLF17A

martukas, 2020-07-20:
These are air filters, not oxygen filters. We need a separate ticket for documenting the oxygen limb of the pneumatic system.

inceptionev, 2020-08-01:
started here: https://github.com/RespiraWorks/Ventilator/tree/inceptionev/add_filter_instructions/manufacturing/alpha-build-instructions/filter-holder

martukas, 2020-08-15:
This is partly done with this:
https://github.com/RespiraWorks/Ventilator/pull/862

I believe the outside clips are still missing.

martukas, 2020-09-01:
The following two items need to be added next:
* the "cross-shaped cover" for inlet filters at positions 1 and 2 on the enclosure (where pipes do not connect).
* link to the OnShape source of the CAD, from which the step files were generated

martukas, 2020-09-01:
A lot of the CAD content is in #657
A ticket is not the right place to store design data! Any meaningful content from that ticket should be migrated into the repository directory structure.

martukas, 2020-09-02:
Missing components added in #895

martukas, 2020-11-04:
Page exists. Improvements tracked in separate tickets such as #892 .

closed 2020-11-04 by martukas