RespiraWorks/Ventilator #803: Update requirements doc files from Valispace

gracecanfield, 2020-07-18:


martukas, 2020-12-18:
We might instead want to migrate all requirements out of Valispace, since no one loves it and we want things as openly accessible as possible

martukas, 2020-12-18:
Which is something we will want to do after #250 gets merged.

dheater, 2021-02-06:
Relates to #52

dheater, 2021-03-21:
@martukas @inceptionev Do we want to migrate the [Covent-19 Specification](https://covent-19.valispace.com/project/34/specifications/groups/13/requirements) or are they obsolete?

martukas, 2021-03-21:
yeah, we don't care about that specification anymore

dheater, 2021-03-22:
I imported them earlier today since they were linked to the system requirements, but we can pull them out if that's preferable.  Let's discuss in the PR #1086

PR #1086 (2021-04-05): 803:Migrate from valispace
# Description

Migrate requirements from Valispace into Sphinx generated documents.

Fixes #803 

Updates #1082

Creates #1084, #1085, #1087, and #1088

Note: These requirements are imported from Valispace. I make no certification that they are correct. If you see issues, please open a new ticket against the requirement.

Note that I did not pull deleted/archived requirements from Valispace. This 
# General checklist:

- [x] I have performed a self-review, including - looked through the `Files changed` tab, browsed repository in my branch
- [x] I have made correspond [... cut]
closed 2021-04-05 by dheater