RespiraWorks/Ventilator #808: Add Oxygen Sensor to Controller Code

inceptionev, 2020-07-20:
Basic requirements:
- Analog input similar to pressure sensors, but on pin PC3
- Should have a linear y= m*x + b conversion between counts and an oxygen concentration [0 - 1.0]
- Output should be callable in the same way as the other sensors.

martukas, 2020-07-23:
Someone could help here by finding out which existing code most resembles what is needed.

inceptionev, 2020-07-24:
Oxygen sensor calibration narrative, assuming amplifier gain g_amp and sensor gain g_o2sensor:

- Nominal values:
  - g_amp: 50 V/V
  - g_o2sensor: 0.06 V/fio2 (where fio2 is 0.0 to 1.0)

- Ventilator turns on, starts self-test
- Run blower for 3 seconds with both inhale and exhale valve fully open (and O2 PSOL fully closed) to flush with ambient air.
- Wait at least 10s.  Can run other self-test procedures, but only using ambient air.
- Record the ambient air sensor value in sensors_zero_vals_[FIO2]
- Read the ambient barometric pressure sensor and record the valve (this is not yet supported in hardware, so for now assume 101.3kpa)
- Oxygen concentration output expressed as a 0.0-1.0 value is:
 - std::clamp( (fio2.voltage() - sensors_zero_vals_[FIO2]) * (101.3/p_ambient_kpa/g_amp/g_o2sensor) + 0.21f)

I have started a sample implementation here if it helps, but note that I am not a software engineer, so this definitely needs to be screened for compliance to coding standards and correct implementation.  However a diff between the current master and this branch can help with finding all the places were modifications are needed to add the extra analog channel: https://github.com/RespiraWorks/Ventilator/tree/inceptionev/add_fio2_sensor

inceptionev, 2020-07-27:
Sample implementation in branch https://github.com/RespiraWorks/Ventilator/tree/inceptionev/add_fio2_sensor now includes the scaling

asmodai27, 2020-07-29:
Working on this right now, I have completed the basic framework and adding some unit tests.

I wonder whether we are using the correct correction factor (101.3 / ambient pressure) to go from partial pressure to fraction of volume...
I am not well versed in how that sensor works, but it seems to me that patient pressure should come into play at some point, as the oxygen sensor is seeing patient pressure.
Any insights on this?

inceptionev, 2020-07-29:
Thanks for picking this up.

That's a good point.  The oxygen sensor is an electrochemical cell, so it actually measuring partial pressure of oxygen, not concentration.  It has a sensitivity of 0.00061V/kpaH20_partial_pressure.

While it is true that the sensor is seeing patient pressure, the ambient pressure has a larger effect.  At 10000ft / 305m, there is a drop of 322cmH2O vs sea level, while the max patient pressure is 35cmH2O.  We could take into account patient pressure as well, but we definitely need to make a correction for ambient pressure.

inceptionev, 2020-07-29:
The sensor has a response time on the order of 10 seconds, so I'm not sure how a quickly-changing patient pressure correction would play out.

asmodai27, 2020-07-29:
OK, no problem there.

Regarding the changes in patient pressure, we could approximate the response time with a similar 10 seconds low-pass filter, but without the actual dynamic of the sensor (especially in terms of phase shift), I am not sure of the actual benefits that this would bring. For now, ambient pressure will suffice.

closed 2020-08-12 by martukas