RespiraWorks/Ventilator #810: Document beta build enclusure

martukas, 2020-07-20:
**Folder/Page:**
There should be a page for the beta build enclosure. Somewhere under beta, I guess.

**Task:**
Get this info from Randall and Tijani and Donna, possibly? some of this was discussed in one of the #eng- channels.
There were discussions about what angle things should be bent at and how that informed selection of manufacturer. Any of this stuff that impacts knowing how to properly get it made should be somehow documented. What sort of machine is needed for cutting the acrylic stuff? Maybe links to some companies that are good candidates for getting it done? But fundamental criteria for selection would be better.

**Subtasks:**
* [ ] CAD files for sheet metal stuff
* [ ] CAD files for acrylic stuff

**Relevant files:**
Probably exported from Fusion or whatever they are using.

martukas, 2021-01-18:
Not sure if this was broken out into sub-tasks, but this probably entails just migrating the CAD and drawings for metal and acrylic to Inventor?

tijanim, 2021-01-19:
This is now contained in #937

tijanim, 2021-01-20:
and somewhat by drawings done in #943

martukas, 2021-02-15:
Superseded by other tickets mentioned above.

closed 2021-02-15 by martukas