RespiraWorks/Ventilator #811: Pet watchdog is being triggered from timer interrupt, a potential issue

subhashr2, 2020-07-21:
While Navigating in code we noticed that "  // Pet the watchdog" is being triggered in "static void high_priority_task(void *arg) {" (which is in-turn a function call form timer interrupt).
If "static void background_loop() {" hangs for any reason, watchdog will never restart the controller board.

martukas, 2020-07-23:
@subhashr2 can you please provide more detail?

martukas, 2020-07-23:
@asmodai27 thoughts on this?

subhashr2, 2020-07-24:
1. As per the comment in main.cpp, "static void high_priority_task(void *arg) {"  function is being called from timer interrupt &
   // Pet the watchdog
  Hal.watchdog_handler(); is being called in this function.
2. If by any chance background_loop hangs at some code location, timer interrupt will never hang as it is being triggered from interrupt handler.
3. Our intention for watchdog is to reset the controller if code hangs some where (this functionality will not be achieved if watchdog is pet from timer interrupt)

asmodai27, 2020-07-28:
Exactly what @subhashr2 said.

We do need a way to check that neither the background loop nor the high priority task are stuck.

Right now the controller itself only checks that the high priority task runs, and resets itself in case of a failure.

I think there is a plan for the RPi to detect an irresponsive controller, which is caused by the controller being stuck in either the high priority task or the background loop, but I am not sure whether it is already implemented, or that the RPi has any means of resetting the STM32.

AFAICT, the high priority task being stuck is less likely, as it has no unbounded loop to hang in, but can cause more damage: while the background loop being stuck would "only" cut out communication with the RPi (making it impossible to change the settings and see the sensor readings), a hanging high priority task would cause erratic behaviour from the actuators in addition to preventing communication to the RPi.