RespiraWorks/Ventilator #815: Risk: Inlet Filter Drops Blower Performance

inceptionev, 2020-07-22:
We need to do a characterization test to demonstrate that the ventilator has acceptable performance with the filters in the flow path.

In particular, the inlet filter is a concern because we have not had a proper inlet assembly to adapt to a filter until now.

Task: use the PAPR inlet assembly to add a filter to the inlet and run the CoVent test set and record the data.

HariGanesh-27, 2020-07-23:
Alternatively, I propose that we use the same filter housing (3D printed filter housing) to do the tests and conclude on performance changes.

inceptionev, 2020-07-23:
I can use the same filter housing, I just need the inlet assembly from the PAPR to connect it to the blower inlet.

inceptionev, 2020-08-30:
I ran a test with all 4 filters installed and little impact on performance was observed, and pip targets up to 35cm were achieved on 1s i-time and 50ml/cmH2O compliance.  The immediate risk of a drastic change in performance is retired.  At a future date, when the full enclosed vent is built, a new baseline performance dataset will be generated.

Ticket OK to close.

closed 2020-08-30 by inceptionev