RespiraWorks/Ventilator #818: Test UFlow PSOL

martukas, 2020-07-23:
**What has to be done**
We must test the UFlow PSOL's that are available for purchase in India. No code is provided for operating them in PWM mode, so this must be tested.

**How do you know it has to be done**
Because these are potential valves to be used in the manufactured vents in India.

**Code pointers**
Currently, there are no integration tests for any PSOL on our system. There are some tests for the binary solenoids at
`software/controller/integration_tests/solenoid_test.h` which could be a starting point.

**Other**
Data sheets have been posted in Slack:
https://respiraworks.slack.com/archives/C0100AC23MJ/p1595483816209200

martukas, 2020-07-23:
Probably blocked by #819

martukas, 2020-07-29:
Might be helped by #838

DonnaCG, 2020-07-30:
Need a software assignee