RespiraWorks/Ventilator #820: Issue BOM to Git

DonnaCG, 2020-07-23:
**What has to be done**
After Beta Build is completed, BOM should be review and updated according to findings from Beta Build.
Final list to be issued to Git for revision control.

martukas, 2021-01-18:
I think most of the BOM is in various sub-assembly pages, so this ticket might be re-scoped as either duplicating that as a complete all-inclusive BOM or just auditing the various pages to make sure nothing is missing.

tijanim, 2021-01-19:
This is now contained in #937

martukas, 2021-02-17:
#1035 ensures that all salient parts are listed in sub-assembly pages. A complete BOM may still have to be generated, but this will be the outcome of a more permanent and automated solution for BOM management as outlined in #1011

closed 2021-02-17 by martukas