RespiraWorks/Ventilator #821: Review Parts Inventory - preparation for beta build

DonnaCG, 2020-07-23:
**What has to be done**
Review Pizza Build and identify current available inventory.
Purpose of this exercise, is to identify missing parts to be ordered in preparation of Beta Build

**How do you know it has to be done**
All inventory columns are completed.

If the quantity identified in "needed" column is not correct, or not precise enough, just update accordingly.
https://drive.google.com/file/d/1vqoaHk_54RKiV9V2NL35gHngVxuSySWt/view?usp=sharing

martukas, 2020-07-27:
This looks like a duplicate of #834

inceptionev, 2020-07-28:
I think it's one for your inventory and one for mine, @martukas

DonnaCG, 2020-07-28:
Yes I created two, one for each build.  But #775 is a duplicate.

inceptionev, 2020-08-13:
Inventory complete

closed 2021-01-18 by martukas