RespiraWorks/Ventilator #828: Implemet pressure support mode

tapemaster, 2020-07-23:
**What has to be done**
Implement a new mode that is referred in requirements as "Presusre Support".
Add it to the GUI.

**How do you know it has to be done**
It was discussed during leads meeting that we need this mode. Captured in the notes here: https://docs.google.com/document/d/1TCQ1UAXR5yLN6OCmcsQ7W-XpVtFkYEnrFRmJU_Qi958/edit#

**Don't forget**
- [x] Put the issue in a milestone
- [x] Assign it to one of the projects
- [x] Add appropriate labels
- [ ] Mention dependent/blocked issues if any

martukas, 2020-07-27:
This is probably defined here: https://github.com/RespiraWorks/Ventilator/blob/master/software/design/ventilation_modes.md
We should probably reference that definition to some requirements docs.

martukas, 2021-04-07:
This is probably implemented, but we need to audit it's compliance with reqs.

asmodai27, 2022-06-02:
Pressure support is implemented in controller, but the placeholder in GUI seems not to activate that mode.