RespiraWorks/Ventilator #829: Implement PIP detection

martukas, 2020-07-23:
Please take the time to describe the issue in detail, so that other contributors can pick it up and implement or declare it done/obsolete in case if was completed out of band.

**What has to be done**
Describe what has to be done, ideally in declarative form - what does success look like.

**How do you know it has to be done**
Link any documents, Slack threads, quote personal communication, reference your personal domain expertise etc. explaining why you think this should be done.

**Code pointers**
Give some thoughts on roughly where in code do you think the necessary changes might be located.

**Don't forget**
- [ ] Put the issue in a milestone
- [ ] Assign it to one of the projects
- [ ] Add appropriate labels
- [ ] Mention dependent/blocked issues if any

inceptionev, 2021-04-07:
This may be helpful.  

- PIP includes the resistive component
- This should include some filtering to avoid capturing noise as the maximum value.

[image]

asmodai27, 2021-04-07:
Code Pointers: check the `BreathSignals` class in GUI: right now the displayed PIP is the maximum pressure attained the previous breath