RespiraWorks/Ventilator #830: Implement PEEP detection

martukas, 2020-07-23:
Please take the time to describe the issue in detail, so that other contributors can pick it up and implement or declare it done/obsolete in case if was completed out of band.

**What has to be done**
Describe what has to be done, ideally in declarative form - what does success look like.

**How do you know it has to be done**
Link any documents, Slack threads, quote personal communication, reference your personal domain expertise etc. explaining why you think this should be done.

**Code pointers**
Give some thoughts on roughly where in code do you think the necessary changes might be located.

**Don't forget**
- [ ] Put the issue in a milestone
- [ ] Assign it to one of the projects
- [ ] Add appropriate labels
- [ ] Mention dependent/blocked issues if any

asmodai27, 2021-04-07:
Code Pointers: check the `BreathSignals` class in GUI: right now the displayed PEEP is the minimum pressure attained the previous breath

inceptionev, 2021-04-07:
This might be useful: 
[image]