RespiraWorks/Ventilator #834: Review Parts Inventory - preparation for beta build

DonnaCG, 2020-07-26:
What has to be done
Review Pizza Build and identify current available inventory with Martin.
Purpose of this exercise, is to identify missing parts to be ordered in preparation of Beta Build

How do you know it has to be done
All inventory columns are completed.

If the quantity identified in "needed" column is not correct, or not precise enough, just update accordingly.
https://drive.google.com/file/d/1vqoaHk_54RKiV9V2NL35gHngVxuSySWt/view?usp=sharing

martukas, 2020-08-12:
I have filled out my column.

I have marked some things in green, meaning **looks like I have enough, but would like a preemptive restock on these**. These are mostly consumables like crimps and connector housings in wiring.

I have marked some things in red, meaning one of the following:
* looks questionable
* looks obsolete, i.e. does not reflect current iteration

For the former, we'll discuss this in Friday's meeting.

For the latter -- this may be indicative that the whole sub-assembly BOM is out of date with what's in github. I have not compared these in detail. If you have bandwidth, please review the relevant pages and see if things look obviously out of sync. Otherwise, we can also talk about this in the meeting.

DonnaCG, 2020-08-18:
I have updated file to indicate what needs to be ordered and where.  Ive also hid all non applicable assemblies.  Items to be ordered are in green.  Please check every tab.

martukas, 2021-01-18:
Most of this has been reviewed and documentation on parts sourcing has been updated in the repo, pending more details on the oxygen branch of the pneumatic circuit.

closed 2021-01-18 by martukas