RespiraWorks/Ventilator #841: Add EEPROM to PCB

inceptionev, 2020-07-29:
This is the hardware side of ticket #712 

Task:
- Select and add to the design an i2c EEPROM for persistent/nonvolatile storage
- Provide a hardware i2c debug port for external access for this EEPROM for testing

Required information:
- Software to confirm required size and speed of the EEPROM

This is required for the next rev of the pcb, however a small engineering PCB may be made to add this to the Rev 1.0 PCB for purposes of accelerating development.

Current progress:
- EEPROM 24LC256T-I/MS currently selected and present in the Rev A PCB design but needs to be confirmed as meeting the need from #712
- i2c debug port not included yet

martukas, 2020-08-07:
What orders of magnitude are our options for size?
I figure we don't need much, even under 1kb might be enough :D 

Here are the sizes of the comms protocol messages:
```
/* Maximum encoded size of messages (where known) */
#define GuiStatus_size                           55
#define ControllerStatus_size                    108
#define VentParams_size                          42
#define SensorsProto_size                        41
```

In addition to that we might want to store valve cycle/wear info, but very likely that will be outsourced to the GUI side. 

The only other usage for non-volatile controller memory could be to record test data without resorting to comms protocol or debug interface to later retrieve this to compare against expected results, or something like that for integration tests.

@asmodai27 @Miceuz other ideas or insights?

martukas, 2021-01-18:
Related to #712

dcstraney, 2021-01-19:
Added to design in #947, can close when confirmed that 24LC256 is adequate for software needs.
Related to #747.

asmodai27, 2021-02-10:
Just merged PR #995 that adds support for reading/writing to 24LC256. Remains to be seen whether that is enough memory for future use. It is sufficient for the uses we are planning at the moment.

inceptionev, 2021-04-26:
implemented by #1101

closed 2021-04-26 by inceptionev