RespiraWorks/Ventilator #842: Include untested files in coverage stats

martukas, 2020-07-29:
**What has to be done**
Test coverage statistics should include files that are untested.

**How do you know it has to be done**
Because our statistics are skewed as untested code is not counted in the total code.

**Code pointers**
@twslankard has worked on this, but seems hard to do. Might need to create dummy functions in those files and explicitly run them, to force the lcov tool to count those files.

**Don't forget**
- [ ] Put the issue in a milestone
- [ ] Assign it to one of the projects
- [ ] Add appropriate labels
- [ ] Mention dependent/blocked issues if any

martukas, 2020-07-29:
Derived from #31

twslankard, 2020-07-29:
Just a note on what I've tried so far...

I was able to add the necessary build flags to generate coverage information for the `native` target. I haven't attempted to add the flags to any other targets because we don't want to link the coverage lib into those. (We could try to create an alternative target that includes the coverage lib, using the STM32 platform, but my guess is that the coverage lib requires linking against the std lib and that is probably a show-stopper. (Correct me if I'm wrong, embedded folks.)

The coverage info generated by the `native` target can be compiled into a report using `lcov` or `gcovr`. Both work fine. However, not all files are included. My hypothesis is that they won't be because no code is present in those files for the `native` target due to the code being excluded by the preprocessor.

For example, all code in the file `controller/lib/hal/flash.cpp` is wrapped by an `if` statement that excludes all code for the `native` target (ostensibly.)

```
// some comments
#if defined(BARE_STM32)

... some code ...
#endif 
<EOF>
```
This makes me think that no coverage info will be generated because there's no code in the resulting translation unit, and thus no symbols in the resulting object file.

As such, we should consider one of the following approaches going forward:

* We decide whether we can change the code to include some of these excluded functions in the `native` target; or
* we try to use a non-native target for code coverage; or
* we do without comprehensive code coverage reports.

martukas, 2020-08-11:
Now that we have coverage working with CI, I'll just add an update to this.
So, I think we have two categories of code that would be tackled by this ticket:
* plenty of controller code as well as GUI code that is simply not tested, and even testing one line of it would get it added to our statistics. This is code that is unaffected by the excluded translation unit problem described by @twslankard above. This is ready to be tackled by someone who picks up this ticket.
* refactoring will be involved. Code that is affected by the above problem definitely needs refactoring to be testable, but even some that is not will also require refactoring for other reasons. This can also be handled within the scope of this ticket, unless it is s something more drastic. Use your own discretion.

PR #1186 (2022-04-07): Test stubs for most gui classes
# Description

* fixes GUI test coverage. It previously did not count this correctly and it seemed like test coverage was more comprehensive than it actually is
* moves a bunch of function implementations from headers into `cpp` files - this was not done for some small classes that can be viewed as a single "page" 
* Fixed some default initialization style and other warnings
* further refactoring and improving of tests is left for later work

Closes #842 

# General checklist:

- [ ] I have performed a self-review, including - looked through the `Files changed`  [... cut]
closed 2022-04-07 by martukas