RespiraWorks/Ventilator #857: Design a more flexible system for calibration parameter storage

martukas, 2020-08-08:
**What has to be done**
It should be possible to connect components with different specifications, such as proportional solenoids or sensors, and not have to manually hardcode new calibration/linearization constants and recompile. These parameters should either be stored in non-volatile memory, or could also be set using some combination of convenient pre-compiler flags. 

**How do you know it has to be done**
We now have multiple types of PSOLs that we are testing with, which have different calibration values. We have been forced to hardcode changes. We also have different specs of stepper motors, possibly other components down the line. It would be good to have a more sustainable approach to accommodating different hardware.

asmodai27, 2020-08-08:
The way we do this at my day job is by storing calibration data in specific parameters files, alongside the program. We could use parts of @sglow's hack to write this data at the end of the program memory using the debug interface, or write this to an external EEPROM.
During boot, the program needs to read the file and check its consistency (checksum, values' bounds, ...) before using any value from this file.
Problem with this approach for critical software is we need a way to demonstrate that no possible combination of accepted values can lead to problems inside the software, which is test heavy (could be unit tests though).
A way around that can be to only allow a few specific values instead of a range of values, but this kind of defeats the purpose of such a file.

martukas, 2021-01-18:
Blocked by #712

martukas, 2021-04-07:
Scope:
* PID parameters
* pinch valve constants
* PSOL constants
* venturi constants
* oxygen sensor...

inceptionev, 2021-04-07:
I suspect that the PID values should actually be immutable (hard-coded), as the dynamics of the system should not be different from one ventilator to the next, as long as the actuators and sensors are properly calibrated.  It might even be possible that a real assembly mistake in a production ventilator could be masked by control tuning that essentially compensates for something wrong, like a high resistance or a leak in the system.

martukas, 2021-09-13:
@inceptionev no need for a very formal write-up. Just some ideas on how to most usefully define a "calibration" in the code. For example, for he pinch valve, we may draw the border around this chunk of code:

```
  value = std::clamp(value, 0.0f, 1.0f);

  // Number of intervals defined by the table.
  float tbl_len = static_cast<float>(calibration_.data.size() - 1);

  // Convert the input value based on a table
  // used to linearize the pinch valve output
  int n = static_cast<int>(value * tbl_len);
  float f = value * tbl_len - static_cast<float>(n);

  if (n == static_cast<int>(tbl_len))
    value = calibration_.data[n];
  else
    value = calibration_.data[n] + f * (calibration_.data[n + 1] - calibration_.data[n]);

  // Convert the value to an absolute position in deg
  // The motor's zero position is at the home offset
  // which corresponds to fully open (i.e. 100% flow)
  // The valve is closed at a position of -MaxMove;
  float pos = (value - 1.0f) * MaxMove;
```

where `value` is mapped to `pos`. This would define a relationship between %volumetric flow vs. stepper angular position. But the question is, do we want the clamp as part of the abstraction?

Other use cases would be:
PSOL -- would it map flow vs. voltage? Also clamp?
Blower -- any nonlinearity to deal with there? What are the input and output units?
Sensor abstractions already map ADC voltage to appropriate units, but is this adequate? Any nonlinearity to deal with there?

Does it even make sense to have a similar implementation for all these cases? If these functions are sometimes represented by a polynomial mathematical operation, even if it is "good enough", is it still most robustly represented by an empirically constructed curve?

I will very likely first refactor the pinch valve, but it would be good to already make something that could be reused to the benefit of other parts.

martukas, 2021-09-13:
Other related tickets:
* #870 
* #838 
* #633 
* #170

inceptionev, 2021-09-26:
I have a strong preference for all actuators (PSOL, pinch valve, blower) to be mapped from 0.0 to 1.0, zero being fully closed, and 1 being fully open.  The calibration ideally would be such that an input from 0 to 1 yields a linear response in flow through the valve.  This makes control easier.

Clamping should be included in the calibration (both PSOL and pinch valve) and should be such that 0 is the lowest point at which further negative motion does not result in a reduction of flow, and 1.0 is the highest point at which further positive motion does not result in an increase of flow.

For the blower, we don't do any closed loop control around this yet, so it's not necessary at this point.  However, having the capability to assign a calibration curve to the blower would be a nice-to-have, in case we do add control in the future.  For now, if we implement blower calibration, we can just set it to 1:1 for now.

I think it makes sense to have all of these use the same calibration system.  The pinch valve uses an 11-point calibration, right?  we could apply that to all the actuators.

The sensors are a bit trickier.

- Venturi has a polynomial conversion.  If the 11-point calibration could sufficiently capture this polynomial curve, we could convert it over to using the interpolated 11-point system.  This would also make it easier to empirically calibrate.  I would say, keep this as is for now, but let's experiment with this once we have a setup that can physically calibrate the venturis.
- Oxygen sensor: we'll only be able to do a 2-point calibration since we can only test at 21% and 100% oxygen, but that is no reason we can't represent this as the same 11-point interpolated calibration, so it makes sense to make the oxygen sensor use this calibration as well.  The 11 points will just be linear interpolations between the two points, extending down to 0% and up to 100%.  This is OK because galvanic oxygen sensors are very linear, even as they age.  The error they gain with age is an offset.

closed 2022-06-02 by asmodai27