RespiraWorks/Ventilator #870: Implement oxygen calibration

martukas, 2020-08-15:
**What has to be done**
It must be possible to calibrate the oxygen sensor.

**How do you know it has to be done**
Saw the gui showing 106%, i.e. here: https://github.com/RespiraWorks/Ventilator/pull/864

**Code pointers**
Ideally:
* this setting should probably be saved in EEPROM on controller side.
* it should be possible to calibrate at will from some sort of advanced options menu in the GUI, or may even be part of self-test at bootup?

**Don't forget**
- [ ] Put the issue in a milestone
- [ ] Assign it to one of the projects
- [ ] Add appropriate labels
- [ ] Mention dependent/blocked issues if any

inceptionev, 2020-08-16:
Oxygen calibration narrative:

- Turn on
- Enter calibration mode
- Prompt user to make sure a source of 100% oxygen is connected to the oxygen inlet
- Prompt user to ensure that no patient is connected 
- (If calibration is combined with power on self test, then user should have also been prompted to connect the inhale port directly to the exhale port with a hose)
- Set exhale valve to position 1.0 (full open)
- Set oxygen PSOL to position 0 (full closed)
- Set blower power to 1.0 (full power)
- Set blower inspiratory valve to position 1.0 (full open) and wait 15s
- Capture and store the fio2 sensor reading in volts.  This is the 21% oxygen level.
- Set blower power to 0 (full off)
- Set blower inspiratory valve to position 0 (full closed)
- Open oxygen PSOL to position 1.0 (full open) 
- Verify that oxygen flow sensor shows a flow of at least 30 L/s and wait 5s
- Set oxygen PSOL to 0.5 (half open) for an additional 10s.
- Capture and store the fio2 sensor reading in volts.  This is the 100% oxygen level.
- Apply oxygen calibration as a linear conversion between these two points.

martukas, 2021-01-18:
Blocked by #712